CrossTheRoadElec/Phoenix-api #7: Using Phoenix' ISmartDrivetrain Utilities w/ Standard WPI Commands

roshanr10, 2017-12-20:
In past, we have integrated our drivetrain code and other subsystems for our autonomous commands.

What is the recommended way to integrate the two using the Phoenix Library?

I believe one could either make a wrapper for a WPI Command that implements ILoopable or make a WPI Command that runs an ILoopable internally.

Are there any major advantages to use CTRE's scheduler versus WPI's implementation? 

[Here is Team 4909's Center Gear Code from the 2017 Season](https://github.com/FRCteam4909/2017-Steamworks/blob/master/src/org/usfirst/frc4909/STEAMWORKS/commands/auto/PlaceMiddleGearEncoder.java)

JCaporuscio, 2017-12-20:
It's unlikely that the ISmartDrivetrain systems will be ready for kickoff (it will likely be a mid-season update), so for now it would be recommended to keep using the WPILib Commands and Schedulers.

Once the CTRE objects are ready, making wrappers using either method you mentioned would be fairly low-risk to implement.

The advantage of the CTRE scheduler is the eventual usage of CTRE servo objects instead of the WPI commands.  These servo classes are designed to take advantage of CAN motor controller features and will integrate seamlessly with the scheduler.

roshanr10, 2017-12-21:
I see, that's good to know. Thanks for the insight about the CTRE scheduler.

Will method signatures likely to change from now to the update? I believe a few method signatures have differed between the FRC-Examples repository and the Alpha. Release on your website; just want to inquire if we could continue to work on development and just plug in the updated executable.

Also, is there a roadmap of what will be changed from now until the official release?

JCaporuscio, 2017-12-21:
I believe at this point the API signatures/prototypes aren't changing -
there will probably be a few additions, but the existing function
prototypes shouldn't change.

I don't really have a roadmap to give you, but we'll be putting together a
migration doc for last year -> this year before kickoff.

In the meantime you can check out our documentation repo:
https://github.com/CrossTheRoadElec/Phoenix-Documentation
We're updating it pretty regularly right now - this is going to be our
official source of documentation moving forward.

Jacob Caporuscio
Software/Hardware Engineer
CTR Electronics

closed 2017-12-21 by roshanr10