CrossTheRoadElec/Phoenix-api #10: Split Phoenix into Device Drivers + Phoenix Utilities

JacisNonsense, 2018-01-05:
Phoenix is fleshing out to be a very hefty library, with a bunch of useful components in it, but for the average FRC team this may be a bit overwhelming.

To avoid changing too much, and reducing the results of scope creep, I think it might be wise to release components of the phoenix library (e.g. TalonSRX, Pidgeon, and other drivers) as standalone components so teams can use what they need, as opposed to importing a giant library and using only a fraction of it.

Although Phoenix does have a bunch of useful components, there has been a lot of concern expressed in the FRC community about the size and scope of the library, particularly on the Java side where system resources come at a premium. 

Perhaps writing the phoenix framework such that it depends on a set of device drivers is a better way forward. It has all the advantages of the current system, but also allows teams who don't need everything phoenix offers to have a straightforward way to get a robot running before competition.

This may be a reasonable goal to put on the 2019 roadmap.

roshanr10, 2018-01-05:
I think it may also be worth considering implementing some more of the relevant WPILib Interfaces so the two are more easily interoperable. I saw the new motor controller wrappers, but it still looks like the design is trending towards using **either** WPILib or CTRE Phoenix, for the most part.

There are surely many teams that would benefit greatly from some of the motion profiling features in Phoenix, especially those that've never done PID before. Yet, those are the same teams that would likely struggle to write the wrappers necessary to use the two frameworks side-by-side. Forcing them to use a completely different framework leaves those teams at a disadvantage when they can't easily reference documentation from other teams, a lot of which has been built up extensively over the past few years.

Beyond these teams, a lot of more advanced teams tend to write their own framework over time, and as such, are more tightly integrated into the WPILib ecosystem as well. Perhaps a viable solution to make usage easier would be to implement interfaces in the existing codebase so they're compliant... I'm not seeing a huge difference between ILoopable that can't be done in WPI Commands; doing that itself may make a lot of Phoenix features more accessible to the average FRC team.

ozrien, 2018-08-03:
@roshanr10 The WPI_* classes in Phoenix don't require ILoopable. ILoopable will be used in future higher level objects (at the mechanism level).  I suspect it will be moved to Phoenix-robotics.  So I don't think it is a either WPI/Phoenix. 

Closing issue as I think this is a subset of https://github.com/CrossTheRoadElec/Phoenix-frc-lib/issues/50, which has the proposed new library breakout (FRC/WPI and non-FRC).  I believe the Phoenix-robotics library will address the concern of the OP.

closed 2018-08-03 by ozrien