CrossTheRoadElec/Phoenix-api #12: License is overly restrictive

d235j, 2018-01-07:
The license (at https://github.com/CrossTheRoadElec/Phoenix-frc-lib/blob/3c6d301f8bd7269a97af0fe65e5b7ff2e3f1f6c2/LICENSE) seems to restrict use of the software only to FRC applications. However, the hardware is also useful for non-FRC applications such as college-level robotics competitions.
I noticed that the license text in the binary installer is not restrictive in this way. Can the license text in the repo be updated to match that one?

ozrien, 2018-01-07:
The intent was that it is permissive if either
- using CTR Electronics hardware products
- using roboRIO for FRC (we don't make the roboRIO).
If your using our hardware products (talon, vic, hero, pdp, whatever), its fine.
Also there are several collegiate groups using our hardware/software.

d235j, 2018-01-07:
@ozrien if that's the case it would probably be useful to clarify the wording to clearly state either/or.

auscompgeek, 2018-08-08:
Given #50, it would be great if the license text was clarified to accurately reflect the intention. I'm sure this could very much become a point of confusion again moving forward. Liberal use of "and/or" would probably suffice.

closed 2021-01-13 by JCaporuscio