CrossTheRoadElec/Phoenix-api #13: Pigeon IMU reporting Firmware Version Invalid

scarmain, 2018-01-08:
Forgive me, I am at work and don't have the specific log files/error strings with me.

Setup (on a bench, all 2018 firmware and Phoenix I downloaded yesterday):
RoboRio, 1 TalonSrx (CAN ID 1), 1 Pigeon IMU (connected to TalonSrx with the Gadgeter cable).  Only the Talon is on the CAN bus (I didn't have a PDP).

I had a really simple bench code that I wrote that would poll the yaw rate every 20ms and put it on the SmartDashboard.   That code works fine (can provide snippets if you need), and I see yaw on the dashboard great.

My issue was the driver station would say ~every 5 seconds that the firmware version for the Pigeon IMU was not available.  I went to the web interface, updated to the 0.19 firmware, verified on the RIO Web Interface it had firmware 0.19, seemed to work.  Even after this verification, I would still see the message appearing on the error window ~every 5 seconds (I did clear out the log).

Is this something you guys are seeing?  Do I need to log some stuff tonight to help you?  (I was just thinking, what if my TalonSrx firmware was old, I updated it like 12/20/17 of last year...)

I did just remember I brought my robot laptop to work, and still had a couple windows open, so here are some screenshots I could take away from the bench:
[image]
[image]

Error Text:
ERROR  103  CTR: Firm Vers could not be retrieved. Use Web-based config to check ID and firmware(CRF) version.   Pigeon IMU (Talon 1)
	at Device_LowLevel::CheckFirmVers(int, int, ctre::phoenix::ErrorCode)
	at LowLevelPigeonImu::GetYawPitchRoll(double*)
	at c_PigeonIMU_GetYawPitchRoll
	at Java_com_ctre_phoenix_sensors_PigeonImuJNI_JNI_1GetYawPitchRoll

JCaporuscio, 2018-01-08:
Your Pigeon IMU firmware is out-of-date.

See: https://github.com/CrossTheRoadElec/Phoenix-Documentation#what-do-i-do-when-i-see-errors-in-driver-station

Per the last part of the error message: "Use Web-based config to check ID and firmware(CRF) version."
A self-test in the web-interface would also have told you to update your firmware.

[image]

Per: https://github.com/CrossTheRoadElec/Phoenix-Documentation#crf-firmware-versions

Version 0.41 is the latest Pigeon IMU firmware.

scarmain, 2018-01-08:
I did remember seeing the FIRMWARE MUST BE UPDATED on the main page of the Pigeon when it was at 0.17.  When I went to 0.19, the error fixed itself.

Where can I find version 0.41 of the firmware?  When I go to the official website: http://www.ctr-electronics.com/gadgeteer-imu-module-pigeon.html#product_tabs_technical_resources , it only shows version 0.19...

scarmain, 2018-01-08:
Guess I should read your documentation better ;) It says it should have been installed on my computer when I installed the Phoenix framework.  I looked at that computer, and I see the new firmware files.

Sorry to have wasted your time.  You can close this issue if you are good too.

closed 2018-01-08 by JCaporuscio