CrossTheRoadElec/Phoenix-api #15: Add getters for persistent configuration parameters

jhh, 2018-01-11:
We're rolling in the 2018 library changes to our graphing and tuning utilities and have been running into API calls that we used to use but are no longer available. An incomplete list includes getters for:

- P, I, D, F
- Forward/Reverse Soft Limits
- Analog Position

In general, in the context of these tools, it would be very useful for us to have a getter for any value that's stored persistently.

JCaporuscio, 2018-01-11:
Given the infrequency of use (and their availability for viewing in the web-interface), most of the individual persistent getter functions were deemed non-critical.

I can understand, though, why it's useful to be able to poll these values - we're not opposed to adding them in the future for the parameters that are used the most frequently.

You still have the ability to get these values through the API, though - similar to last year, there's a function (configGetParameter, documented in the migration guide) that can be used to get any parameter that's in the enum list (it's extensive).  The "ordinal" argument for this function depends on the parameter, but for slot data (PIDF etc.) it's the slot index you want to use.

So getting the D value from profile slot 0 (with a 10ms timeout) would look like this:
talon.configGetParameter(ParamEnum.eProfileParamSlot_D, 0, 10);

On the note of Analog Position, the same function from last year is still present, but like the rest of the specific getXXXXPosition functions, it was placed under the getSensorCollection() call (documented in the migration guide).

jhh, 2018-01-11:
Thanks, I'll take a look at configGetParameter.

Looking again at  the Javadoc, I do see that `getAnalogIn` reports position. It's a minor point, but in a future release you might consider deprecating this method and adding `getAnalogInPosition`

> like the rest of the specific getXXXXPosition functions

jhh, 2018-09-24:
Looks like what I needed was added in 5.7  `TalonSRXConfiguration` and related - good stuff! I'll leave this open in case you were tracking something else with this issue otherwise it can be closed.

ozrien, 2018-09-24:
> Looks like what I needed was added in 5.7 `TalonSRXConfiguration` and related - good stuff! I'll leave this open in case you were tracking something else with this issue otherwise it can be closed.

Also related kinda - check out https://github.com/CrossTheRoadElec/Phoenix-diagnostics-client.
This is related to configs in that it will replace the RIO web-based config.  I think strikeforce does something similar where you have config files that you apply via scripts.
We switched to JSON, and our server binary loads them up.  You control it via HTTP API, see open-source C# client for examples.

It's not updated yet with all the configs (has same level of config support as last year's web-based config + self-test).

closed 2021-01-13 by JCaporuscio