CrossTheRoadElec/Phoenix-api #16: CAN frame not received/too-stale

bf2799, 2018-01-12:
We have a Pigeon IMU connected to a CAN Talon SRX. Whenever we try to interface with our Pigeon through robot code, it gives a `CAN frame not received/too-stale` error.

For example, with this code:
```
package org.usfirst.frc.team1768.robot;

import edu.wpi.first.wpilibj.IterativeRobot;
import com.ctre.phoenix.sensors.PigeonIMU;

public class Robot extends IterativeRobot {

	public void robotInit() {
		PigeonIMU pigeon = new PigeonIMU(3);
		System.out.println(pigeon.getState());
	}
}
```

it prints that the state is "NoComm" and it gives the following errors:

```
ERROR  -3  CTR: CAN frame not received/too-stale.  Pigeon IMU 3 Get State
	at Java_com_ctre_phoenix_sensors_PigeonImuJNI_JNI_1GetState

ERROR  -3  CTR: CAN frame not received/too-stale.  Pigeon IMU 3 GetState
com.ctre.phoenix.sensors.PigeonImuJNI.JNI_GetState(Native Method)
com.ctre.phoenix.sensors.PigeonIMU.getState(PigeonIMU.java:547)
org.usfirst.frc.team1768.robot.Robot.robotInit(Robot.java:30)
edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:31)
edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:250)
```

Our Pigeon is connected to the correct Talon SRX (shown by the Talon self test and lighting up the LEDs on the Talon), and the Pigeon and Talon are up to date (Pigeon firmware 0.41, Talon SRX firmware 3.1). The CAN bus utilization isn't too high (it's at roughly 25%). The Pigeon self test doesn't show any obvious issues and reads out good values for everything - for example, the yaw is accurate.

The Pigeon was working successfully before we updated to the 2018 software.

JCaporuscio, 2018-01-12:
If your Pigeon IMU is connected to a Talon SRX, you need to be constructing the pigeon object using the appropriate Talon SRX object, not an ID.

This would construct a Pigeon object for a device connected to the CAN bus:
PigeonIMU pigeon = new PigeonIMU(3);

This will construct a Pigeon object for a device plugged into a Talon SRX:
TalonSRX talonConnectedToPigeon = new TalonSRX(DeviceID);
PigeonIMU pigeon = new PigeonIMU(talonConnectedToPigeon);

This is documented in the API docs:
[image]

If these are not visible to you within Eclipse, follow the instructions on attaching your Java source:
https://github.com/CrossTheRoadElec/Phoenix-Documentation#add-javadoc-if-using-java

This is the same way you needed to construct Pigeon IMU objects last year - this has not changed.

JCaporuscio, 2018-01-13:
Were you able to resolve your issue?

JCaporuscio, 2018-01-15:
Closed due to no response.

closed 2018-01-15 by JCaporuscio