CrossTheRoadElec/Phoenix-api #18: pushMotionProfileTrajectory throws a null pointer exception if TrajectoryPoint.timeDur is not set

NoahGleason, 2018-01-15:
It should probably default to 0 instead of throwing a NPE in a separate thread and causing the Talon to not execute the profile.

ozrien, 2018-01-15:
Generally you should fill every member of TrajectoryPoint to ensure proper operation.

Since a null timeDir is not a sensible value, throwing is probably best.   Although ideally the TrajectoryPoint constructor should fill all the values so the push routine doesn't throw if you create a new TP, and only fill the members you care about.  

However, I'm not sure what timeDur should default to, the ideal default value depends on whether you are using the base trajectory duration feature or not.

Regardless you should tell the software how fast you want to run your points or you will not get the result you expected.

ozrien, 2018-08-03:
Issue added to our private tracker.

JCaporuscio, 2021-01-13:
This has been fixed since 5.8.0

closed 2021-01-13 by JCaporuscio