CrossTheRoadElec/Phoenix-api #19: Soft Limits Don't Work

rgreenblatt, 2018-01-17:
After trying and failing to get soft limits to work on a mechanism controlled with positional PID, we tried setting the forward and reverse limits to 0 (with soft limits enabled, using the web dash). The motor's movement was still completely unrestricted. We were using a CTRE_MagEncoder_Absolute with PulseWidthEncodedPosition = 8.

ozrien, 2018-01-17:
Make sure in robot-code you have not overridden-OFF the soft limits.  Pass true to OverrideSoftLimitsEnable() if you are using this function. 

Also be sure to Config the Threshold and Enable for the sides you care about...

These are the relevent functions...
ConfigForwardSoftLimitThreshold (int forwardSensorLimit, int timeoutMs) 
ConfigReverseSoftLimitThreshold (int reverseSensorLimit, int timeoutMs) 
ConfigForwardSoftLimitEnable (bool enable, int timeoutMs) 
 ConfigReverseSoftLimitEnable (bool enable, int timeoutMs) 
OverrideSoftLimitsEnable (bool enable)
[C++ API](http://www.ctr-electronics.com/downloads/api/cpp/html/classctre_1_1phoenix_1_1motorcontrol_1_1can_1_1_base_motor_controller.html)

Also use the web-based config to confirm the checkbox and thresholds for soft-limit.

Make sure the sensor is in phase, once you've done that ,you can SetInverted() to choose what you want to be "forward" and "reverse".  If the sensor is not in phase, the feature will not function correctly.

ozrien, 2018-01-24:
Sounds like the issue was resolved based on an offline conversation.  So closing issue.

closed 2018-01-24 by ozrien