CrossTheRoadElec/Phoenix-api #26: getOutputCurrent() is in BaseMotorController but presumably not supported by SPX

NoahGleason, 2018-01-26:
According to https://content.vexrobotics.com/vexpro/pdf/VictorSPX-InfoSheet-20171129.pdf, the Victor SPX can't measure current, but it still has the getOutputCurrent() method it inherits from BaseMotorController. This should probably be moved to TalonSRX.

ozrien, 2018-01-26:
This is correct and I agree. In the meantime I expect the return is zero amps when used with the Victor SPX.

aswerdlow935, 2018-02-20:
Yes, the return is zero.....only saw this after testing for days with the method only to later realize it was always returning zero. Should definitely be moved (or at the least have a warning put in for now).

ozrien, 2018-02-20:
Depending on what you are trying to accomplish, you can poll the current via PDP.

JCaporuscio, 2021-01-13:
This has been fixed since 5.8.0

closed 2021-01-13 by JCaporuscio