CrossTheRoadElec/Phoenix-api #32: PigeonIMU setYaw using incorrect units

camearle20, 2018-02-08:
It would appear that the setYaw (and possibly the other setters) are not actually using degrees (as is specified by the parameter name 'angleDeg').  Instead, some unit mapping to 64 units per degree appears to be in place.  To set angle to 90 degrees, we have to call `setYaw(5760)`.  Calling `setYaw(90)` results in an angle of about 1.4 degrees being set.  The Javadoc also specifies degrees, although the "max" value specified aligns with this mapping of 64 units per degree (23040 / 360 = 64).
`@param angleDeg  Degree of Yaw [+/- 23040 degrees]`

For reference, we are running the latest Phoenix version (5.2.1.1) and the latest Pigeon firmware (0.41)

allengregoryiv, 2018-02-22:
Is this issue fixed in 5.2.2?

JCaporuscio, 2018-02-26:
This is documented in the errata: https://github.com/CrossTheRoadElec/Phoenix-Documentation/blob/master/README.md#pigeon-imu-set-routines-divide-input-by-64

To check the changes for a given release, look at the release notes.  You can always view the latest release notes from our website.  Direct link: https://www.ctr-electronics.com/downloads/release_notes/RELEASE_NOTES

This has not been fixed in 5.2.2 and will likely not be fixed during the FRC season.  As much as possible, we do not change function behavior during the season.

JCaporuscio, 2019-06-15:
This was fixed before the 2019 season.
Closing tracker since the public API repo is now up-to-date with the current release.

closed 2019-06-15 by JCaporuscio