CrossTheRoadElec/Phoenix-api #39: Need separate AnalogPotentiometer and AnalogEncoder feedback devices

pydude, 2018-04-06:
This is about the behavior of the Analog feedback device during position closed loop control on a Talon SRX.

In some earlier versions of Phoenix, there existed feedback devices AnalogPot and AnalogEncoder. This needs to be the case in the current version. In the current version, there is only Analog. 

The position reading is increased by 1023 anytime that the Talon thinks that the analog device has wrapped around. This is the desired behavior for Analog Encoders.

However, when using a potentiometer, this can cause issues. Sometimes, a brief electrical short can occur in between the potentiometer power leads. The Talon then reads this as a wrap-around, changing the value by 1023. This causes the position to be incorrect until the next time the Talon starts up. What would be a brief bad reading becomes a long-lasting one. The Talon sees that it is off from its target position by a lot, and therefore tries to correct it. This causes it to constantly try to move in one direction to try to reach an unreachable position.

In a perfect robot, this problem would never occur. However, sometimes things aren't perfect. There needs to be a way to disable wrap-around on position readings, to prevent a minor issue from completely wrecking the position control.

ozrien, 2018-04-07:
This is addressed already in the Talon SRX / Victor SPX Software reference manual...
[image]

https://github.com/CrossTheRoadElec/Phoenix-Documentation/raw/master/Talon%20SRX%20Victor%20SPX%20-%20Software%20Reference%20Manual.pdf

pydude, 2018-04-07:
Oh... I guess I did not see it because it did not have a specific function to set it. I did not realize that some settings can only be changed with configSetParameter. 

Well, good to see that there is a solution. Thanks for the help!

closed 2018-04-07 by pydude