CrossTheRoadElec/Phoenix-api #49: New Feature: Motion Profile Arc With Feed Forward

ozrien, 2018-08-03:
The current motion profile arc has the following limitations...
- the auxiliary closed-loop has a target instance variable for position but no feed-forward/target-velocity member.
- the primary closed-loop has a target velocity (implicit feed-forward) which is generally used for straight (translational) motion.
- the feed-forward that exists is primarily based on a velocity feedforward kF (also referred to as kV).
Ideally it would be possible to specify many terms (velocity, accel, static, whatever) or just provide arbitrary motor power and use whatever model you want.

Proposed changes
- If we change the traj point member from PID[0] target-velocity to PID[0] arbitrary-feedfwd, that will allow for straightforward implem of kV, kA, kS.  This will also free up space in the framing (target velocity requires more bits than simple motor percent).
- The above change allows space for an second auxiliary-arbitrary feedfwd.  So for example, if you are using ProfileArc for a differential drive, you now have a feedfwd for PID[0] (the straight one), and a feedfwd for aux PID[1] (the turning one).
- This strategy requires API level implementation of velocity-feedforward instead of doing it in the Talon SRX/ Victor SPX firmware.  This could mean the kF term in the firmware is not what's used, or API would have to synchronously track the kF config so it uses the correct gain.
- To support the legacy use-case, a new ControlMode might be added.  Trajectory Point structure will likely need to hold both old and new members (gross) or we have a unique trajectory structure for each control mode (also gross).  Additionally API could yell at the user if the wrong members are filled with nonzero values.

Jared341, 2018-08-03:
+1 this would be super useful.

JCaporuscio, 2019-06-14:
Motion Profiling was updated this past FRC season to include this feature.

Given that this public API repository is now up-to-date with this feature, I'm closing this issue tracker.
Relevant (Java) Trajectory Point member can be seen here: https://github.com/CrossTheRoadElec/Phoenix-api/blob/master/src/main/java/com/ctre/phoenix/motion/TrajectoryPoint.java#L21

closed 2019-06-14 by JCaporuscio