CrossTheRoadElec/Phoenix-api #54: Phoenix Dependency-WPILib Version Mismatch

copernicium, 2018-08-13:
WPILib has undergone updates and refactoring., but Phoenix has not updated to match. Since Phoenix depends on WPILib, it cannot be built against either the current version of WPILib (from source) or the most recent release, which also means it can't be used with either.

JCaporuscio, 2018-08-13:
Is there a specific issue you're running into?

You are correct that Phoenix cannot be built against current WPILib source, but Phoenix is still compatible with the latest official release of WPILib (currently v2018.4.1).

As the FRC Beta approaches Phoenix will be updated so that it can be built for either latest official WPILib or latest development WPILib.

In the meantime Phoenix can still be built against and used with the official WPILib release.

copernicium, 2018-08-13:
Thank you for your quick reply. 

I was encountering a namespace mismatch/include with llvm, since they moved wpiutil llvm code to a wpi namespace and directory, deprecating llvm. Phoenix's WPI_TalonSRX and WPI_VictorSPX files still reference raw_ostream as part of llvm, causing undefined reference errors when I tried to link and run executables against those libraries.

JCaporuscio, 2018-08-13:
Yep, that's a compatibility issue with the current WPILib library.

If you want to re-build Phoenix to be compatible with latest development WPILib, you can look at their motor controller class implementations and modify the WPI_* classes in your local checkout of Phoenix-frc-lib to match.  Re-building would then give you binaries compatible with new WPILib.  I believe the llvm namespace move is the largest change, but there may be other minor changes necessary to get Phoenix to build against latest WPILib.

Sometime in the next few weeks we'll likely make a dev branch of Phoenix that incorporates these changes, though they won't get merged into master until later.

Otherwise you can continue to use the released version of Phoenix with the official WPILib release.

copernicium, 2018-08-13:
I've tried using patch files to fix Phoenix, but it's been tough to get everything working, especially with WPILib's major refactors (one or two weeks ago, they moved a lot of stuff around). 

Recently, building WPILib's latest release (v2018.4.1) stopped working, and according to one of the developers working on it, they no longer support last season's releases. He was actually surprised that we could ever compile it, since they deprecated it a few months ago. 

I could get Phoenix compiling with some work, but that would complicate our build process and version control system quite a bit. Plus, future updates to both Phoenix and WPILib would break it.

JCaporuscio, 2018-08-13:
Is there a reason you're re-building WPILib as part of your build process?

I can assure you that the following is true:

- Phoenix on master can be built against the 2018.4.1 WPILib release.  This does not require building WPILib as the Phoenix gradle project automatically pulls down the relevant WPILib artifacts during build.  I sanity checked this today and it still works.

- If you use Eclipse, the Release WPILib plug-ins, and the latest Phoenix installer, you will be able to build and deploy your robot project.

Speaking with some of the WPILib developers, it looks like they're having an issue with their maven repository right now that prevents the 2018.4.1 source from being built.  Once this issue is resolved 2018.4.1 should be buildable again.
2018.4.1 is still supported (it's the latest official release).  It's the same version that's being used for the GradleRIO alpha test.

Unfortunately if you'd like to build Phoenix against the latest development version of WPILib you'll have to modify the Phoenix source yourself (at least for the time being).  The easiest way to do this would be to remove the WPI_* classes, provided you don't use them in your robot project.

As I mentioned previously we'll have a branch in the next few weeks that make Phoenix compatible with latest development WPILib, it just isn't something we're working on yet.  Once beta is closer and most of WPILib's changes have been integrated we'll start converting Phoenix.

copernicium, 2018-08-13:
We rebuild Phoenix and WPILib for convenience, since it requires a lot more set-up to download the Phoenix and WPILib libraries (i.e. installing Eclipse and the FRC plugin and the Phoenix LifeBoat) whereas we can build everything (our project included) using a CMake script. 

Anyway, I didn't know WPI was planning on fixing their packages. We'll go back to building Phoenix when they do that. We mostly needed Phoenix for testing purposes, so we can wait for CTRE and WPI to fix their packages/dependencies. 

Thank you for your support.

closed 2018-08-13 by copernicium