CrossTheRoadElec/Phoenix-api #56: Typo of "auxiliary" in config structs

auscompgeek, 2018-11-17:
https://github.com/CrossTheRoadElec/Phoenix-api/blob/5da24d17adc9db5fd2564453a0649254e8834a10/cpp/include/ctre/phoenix/MotorControl/CAN/TalonSRX.h#L44
https://github.com/CrossTheRoadElec/Phoenix-api/blob/5da24d17adc9db5fd2564453a0649254e8834a10/cpp/include/ctre/phoenix/MotorControl/CAN/VictorSPX.h#L35
https://github.com/CrossTheRoadElec/Phoenix-api/blob/5da24d17adc9db5fd2564453a0649254e8834a10/java/src/com/ctre/phoenix/motorcontrol/can/TalonSRXConfiguration.java#L11
https://github.com/CrossTheRoadElec/Phoenix-api/blob/5da24d17adc9db5fd2564453a0649254e8834a10/java/src/com/ctre/phoenix/motorcontrol/can/VictorSPXConfiguration.java#L10

Auxiliary is spelt correctly everywhere else.

JCaporuscio, 2018-11-19:
Good Catch.  We'll get this fixed in an upcoming release.

JCaporuscio, 2018-11-26:
This is now fixed in the latest Beta release (v5.10.0).

closed 2018-11-26 by JCaporuscio

auscompgeek, 2018-11-26:
Is there at source code available for this release anywhere?

JCaporuscio, 2018-11-26:
Not currently as the public repo is out-of-date.
It's on our to-do list to update but I don't have an exact ETA.