CrossTheRoadElec/Phoenix-api #60: No documentation for config_**(int slotIdx, double value) methods

retrodaredevil, 2019-01-07:
In BaseMotorController each config_**(int slotIdx, double value, int timeoutMs) has documentation while each one without a timeoutMs parameter is missing documentation. This makes pulling up the documentation in IDEs slightly more of a hassle. Example: https://github.com/CrossTheRoadElec/Phoenix-api/blob/master/java/src/com/ctre/phoenix/motorcontrol/can/BaseMotorController.java on line 1222

JCaporuscio, 2019-01-07:
I agree that this is an issue.  
This is being worked on and will be corrected in a future release (though no exact eta).

JCaporuscio, 2019-06-14:
As of the update to latest Phoenix version, this should be resolved for all config* functions in both C++ and Java.

Example: https://github.com/CrossTheRoadElec/Phoenix-api/blob/master/src/main/java/com/ctre/phoenix/motorcontrol/can/BaseMotorController.java#L1581

closed 2019-06-14 by JCaporuscio