CrossTheRoadElec/Phoenix-api #62: JVM Crash caused by JNI call in Phoenix-api

VanJoel, 2019-01-30:
We had the JVM crash when running code that had worked for a decent amount of time.

The error is not reproducible as far as we can tell but we wanted to report this in the case that it is useful
Phoenix API v5.12.0
WpiLib v2019.1.1

'''
Last login: Tue Jan 29 18:23:24 2019 from 10.49.10.68
admin@roboRIO-4910-FRC:~# cd /tmp
admin@roboRIO-4910-FRC:/tmp# ls -latr
total 52
drwxrwxrwt    5 admin    administ       100 Dec 31  1969 ../
-rw-rw----    1 lvuser   ni               8 Mar 31  2018 ni_boot_mode_input
drwxrwxrwx    3 admin    administ        60 Mar 31  2018 natinst/
-rw-r--r--    1 admin    administ         0 Mar 31  2018 nipalk-load.log
drwxrwxrwx    2 webserv  ni             100 Mar 31  2018 nipal/
-rw-r--r--    1 lvuser   ni           47362 Jan 29 18:21 hs_err_pid2005.log
drwxr-xr-x    2 lvuser   ni              60 Jan 29 18:29 hsperfdata_lvuser/
drwxrwxr-x    2 lvuser   ni             180 Jan 29 18:29 frc_versions/
drwxrwxrwt    6 admin    administ       180 Jan 29 18:29 ./
admin@roboRIO-4910-FRC:/tmp# cat hs
hs_err_pid2005.log  hsperfdata_lvuser/
admin@roboRIO-4910-FRC:/tmp# cat hs_err_pid2005.log
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGBUS (0x7) at pc=0xab0290c4, pid=2005, tid=2037
#
# JRE version: OpenJDK Runtime Environment (11.0.1.13) (build 11.0.1.13-frc+0-2019-11.0.1u13-1)
# Java VM: OpenJDK Client VM (11.0.1.13-frc+0-2019-11.0.1u13-1, mixed mode, serial gc, linux-)
# Problematic frame:
#
[error occurred during error reporting (printing problematic frame), id 0x7, SIGBUS (0x7) at pc=0xb6ef2d88]

# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -Djava.library.path=/usr/local/frc/third-party/lib /home/lvuser/RobotCode.jar

Host: rev 0 (v7l), 2 cores, 244M, NI Linux Real-Time 6.0
Time: Tue Jan 29 18:21:53 2019 EST elapsed time: 11 seconds (0d 0h 0m 11s)

---------------  T H R E A D  ---------------

Current thread (0xad57fc00):  JavaThread "Notifier" daemon [_thread_in_native, id=2037, stack(0xab060000,0xab0b0000)]

Stack: [0xab060000,0xab0b0000],  sp=0xab0ae958,  free space=314k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)

[error occurred during error reporting (printing native stack), id 0x7, SIGBUS (0x7) at pc=0xb6ef2d88]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  com.ctre.phoenix.motorcontrol.can.MotControllerJNI.GetSelectedSensorPosition(JI)I+0
j  com.ctre.phoenix.motorcontrol.can.BaseMotorController.getSelectedSensorPosition(I)I+5
j  com.team4910.frc2019.subsystems.DriveTrain.readPeriodicInputs()V+27
j  com.team4910.frc2019.subsystems.SubsystemManager$Disabled.run()V+33
j  com.team4910.frc2019.iterator.Iterator$1.runCrashTracked()V+62
j  com.team254.lib.util.CrashTrackingRunnable.run()V+1
j  edu.wpi.first.wpilibj.Notifier.lambda$new$0()V+114
j  edu.wpi.first.wpilibj.Notifier$$Lambda$10.run()V+4
j  java.lang.Thread.run()V+11 java.base@11.0.1.13-frc
v  ~StubRoutines::call_stub

[... 708 lines cut ...]
CPU revision    : 0

processor       : 1
model name      : ARMv7 Processor rev 0 (v7l)
BogoMIPS        : 666.66
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc09
CPU revision    : 0

Hardware        : Xilinx Zynq Platform
Revision        : 0003
Serial          : 0000000000000000

Memory: 4k page, physical 250256k(2952k free), swap 0k(0k free)

vm_info: OpenJDK Client VM (11.0.1.13-frc+0-2019-11.0.1u13-1) for linux--vfp-sflt JRE (11.0.1.13-frc+0-2019-11.0.1u13-1), built on Oct 20 2018 00:04:47 by "" with gcc 6.3.0

END.
admin@roboRIO-4910-FRC:/tmp#
'''

We will report if we figure out how to reproduce this error and if it occurs again.

JCaporuscio, 2019-01-30:
We will investigate, and any additional information/context you can provide will be helpful.

When did this occur during use?  Was it while operating the robot, or during initialization?

ozrien, 2019-01-30:
Just zip up your project and email it to us.  support@ctr-electronics.com

VanJoel, 2019-02-01:
I have sent the related code. 

As for further context I believe that it was shortly after initialization, however I will confirm this once I have a chance to get the logs from that run.

JCaporuscio, 2021-01-13:
Closing as issue is old and has not been reproduced.

closed 2021-01-13 by JCaporuscio