CrossTheRoadElec/Phoenix-api #70: TalonSRX and TalonFX openLoopRamp config effects close looping in velocity

nimrod46, 2020-02-24:
We are using the velocity control mode to control our motor and it seems that defining custom open loop ramp value while having zero in close loop ramp value makes the close loop use the open loop value

We tasted it using TalonSRX and TalonFX, I didn't have much time testing this I just disabled the ramp completely so I don't know what will happened if I define custom value in the close loop value and it's effect if it will be less then the open loop value or if the bug occurs in other close loop modes.

We default all motor using the `configFactoryDefault` method and config it using the `configAllSettings` method.

JCaporuscio, 2020-02-24:
Do you have a set of steps to reliably reproduce the issue you're seeing?
How/what are you measuring that indicates the openLoopRamp setting is improperly affecting closed-loop output?

nimrod46, 2020-02-25:
> Do you have a set of steps to reliably reproduce the issue you're seeing?
> How/what are you measuring that indicates the openLoopRamp setting is improperly affecting closed-loop output?

I don't have access to a robot to test it, you can't reproduce the bug? I used a big ramp to see that it's takes effect, I.e put 10 in the open loop ramp then 0 and saw the difference while close-looping.

JCaporuscio, 2020-02-25:
You have a claim that no one else has reported yet, and a quick sanity check on our end does not reveal the issue you're claiming.

So in order to address the issue we need detailed steps to reproduce the problem.  In general, this is something you should provide whenever reporting an issue (to us or anyone else).

nimrod46, 2020-02-28:
> You have a claim that no one else has reported yet, and a quick sanity check on our end does not reveal the issue you're claiming.
>
> So in order to address the issue we need detailed steps to [...]

I will make a quick test code and share it once I will be able to test it and make sure it's contains the bug I described.

JCaporuscio, 2021-01-13:
Closing since issue has not been reproduced.
If issue is still current, please e-mail support@ctr-electronics.com or make a new tracker on our releases repo.
https://github.com/CrossTheRoadElec/Phoenix-Releases

closed 2021-01-13 by JCaporuscio