CrossTheRoadElec/Phoenix-api #71: `configSupplyCurrentLimit()` should be declared in `IMotorControllerEnhanced` interface and as `abstract` in `BaseTalon`

richiksc, 2020-02-29:
The `TalonFX` has a new API for configuring current limits, `configSupplyCurrentLimit().` I was glad to see that the same API has been added to TalonSRX. However, the method is not declared in the interface itself.

## Why it is beneficial.

Our team stores references to motor controllers as `IMotorControllerEnhanced`. This allows us to easily swap between TalonFX and TalonSRXs, as well as "ghost" motor controllers, which uses a custom implementation of `IMotorControllerEnhanced` which is essentially a no-op so we can disable motors and even remove them physically without breaking robot code or having to significantly refactor. However, after the splitting of the current limit APIs, we can no longer simply current limit a motor without having to a type check and cast, which is unwieldy:

```java
if (motor instanceof TalonFX) {
  ((TalonFX) motor).configSupplyCurrentLimit(new SupplyCurrentLimitConfiguration(...))
} else if (motor instanceof TalonSRX) {
  ((TalonSRX) motor).configSupplyCurrentLimit(new SupplyCurrentLimitConfiguration(...))
} else {
  ...
}

## A series of steps to test the feature after it is implemented

Create a motor variable of type `IMotorControllerEnhanced` and initialize it to either a `TalonFX` or a `TalonSRX`. Call `configSupplyCurrentLimit` without casting it to a subtype.

ozrien, 2020-02-29:
One workaround might be to use BaseTalon references instead of IMotorControllerEnhanced references, which brings in the new configSupplyCurrentLimit, which you can then override in your custom child class.

richiksc, 2020-03-01:
@ozrien Without declaring it in the interface, we will not be able to use the "Ghost Talons" cleanly - which are essentially no-op implementations of `IMotorControllerEnhanced`.

JCaporuscio, 2021-01-13:
This has been fixed with the 2021 Kickoff release (5.19.4.1)

closed 2021-01-13 by JCaporuscio