CrossTheRoadElec/Phoenix-api #73: This repo is severely out of date

PeterJohnson, 2020-08-28:
This repo is more than a year out of date, and does not reflect what is currently hosted on Maven.  This makes it difficult to find reference source code when issues arise.

ItayZiv, 2020-08-28:
It seems like this repo isn’t being updated anymore, and the releases repo (https://github.com/CrossTheRoadElec/Phoenix-Releases) is the place to look at now, if this is the case this repo should be archived after moving all the issues to avoid confusion...

PeterJohnson, 2020-08-28:
As far as I can tell, the releases repo does not contain any browsable source code, only binary releases.  I’m looking for online browsable source code.

retrodaredevil, 2021-01-09:
I'm also looking for source code for what's hosted on Maven. Decompiling in IntelliJ is great and all but...

Is CTRE not publishing their source code anymore?

ozrien, 2021-01-13:
The Phoenix-api repository is out of date.  We have not been very disciplined about updating it for the following reasons...
1. It tends to be an after-thought after the proper/tested Phoenix releases.
2. The API documentation is auto-generated and placed on the website anyway.
3. Comment blocks should auto appear in VSCode regardless.
4. There have been competitor products that have copied our intellectual property and then made FRC-legal.  One particular case resulted in us taking legal action.

The last point makes deciding what to do in the future difficult - I have been given no assurances that this won't happen in the future.

So I spent the last few days thinking about this more.  

In the past we (CTRE) have taken lots of risk in bringing really creative ideas to the market.
Ideas that people usually think is not possible initially.
- Making a battery tester that fits in your pocket (Tests instantly - How is that possible?).
- Releasing the first sealed brushed motor controller (No fans? - How is that possible?)
- Releasing several unique CANbus devices - proving CAN bus can work (really well) in FRC as the back-bone of the controller system - when many believed this would not work.
- Diagnostic server (safely handles talons with the same device ID - how can it do that?).  And supports simulated devices now?
- The first high-performance integrated brushless motor solution in FIRST (with reverse protection and thermal-protection).
- ... and yes even coordinated music playing across Falcons  :)

If CTRE was a company that just manufactures commonly used commodities I might feel differently about this.  
But the reality is we are a company that relies heavily on our intellectual property.

So I think the best step moving forward is to not update this repository.
We will likely just archive the existing repository now for posterity - but it will not be updated.
Issues will be closed or moved to https://github.com/CrossTheRoadElec/Phoenix-Releases/issues.

This might change in the future - for example maybe Phoenix-wpiapi could have viewable source.
 
Funny enough the 2021 Kickoff video mentions teaching FRC students the importance of protecting your own IP (see below).
Keeping with this, I think documenting what our position is and why we have made this decision will help meet this goal.
https://youtu.be/cyHxkWF1KUQ?t=1766
[image]

pietroglyph, 2021-01-13:
Would it be possible to mark the repo as archived and add a notice to the README? In its current state it causes some confusion.

JCaporuscio, 2021-01-13:
Yep, that's the plan.  I can update the readme now, the repo will be archived some time this week.

retrodaredevil, 2021-01-14:
Thank you for the detailed response. That helps a ton.

> This might change in the future - for example maybe Phoenix-wpiapi could have viewable source.

Yeah that's normally the source code that I'm interested in (I mean the java API source is already available, it's just more of a pain to download and browse). It would be nice to be able to have a GitHub repo for that where people could open issues and even create pull requests for simple stuff (maybe a method needs to be public or documentation needs to be updated). It would also be useful to see change history and commit messages for certain files so random people like me can better understand changes such as changing from ints to doubles, and other random stuff like that.